SuperDARNCanada/borealis #13: Got a "No first ack from driver - This shouldn't happen"

kkotyk, 2018-01-16:
State:
Ran 1048 successful sequences until bus error (core dumped) message on signal_processing and n200_driver. source mode release.
At around the same time that the bus errors occurred, the radar_control.py code raised the following:
`RXSEQNUM 1048
TXSEQNUM 1049
New samples 0 True False [0]
New samples 21000 False False [0]
New samples 33000 False False [0]
New samples 36000 False False [0]
New samples 40500 False False [0]
New samples 46500 False False [0]
New samples 63000 False False [0]
New samples 64500 False True [0]
Traceback (most recent call last):
  File "radar_control/radar_control.py", line 482, in <module>
    radar()
  File "radar_control/radar_control.py", line 470, in radar
    poll_timeout, seqnum_start + nave)
  File "radar_control/radar_control.py", line 258, in verify_completed_sequence
    raise ExperimentException(errmsg)
experiments.experiment_exception.ExperimentException: Did not receive ack from either rx_ack_socket or tx_socket`

Then, I restarted signal_processing, n200_driver and radar_control.py but I got the following:
`NEW EXPERIMENT FOUND
NEW EXPERIMENT FOUND
New AveragingPeriod
New samples 0 True False [0]
New samples 21000 False False [0]
New samples 33000 False False [0]
New samples 36000 False False [0]
New samples 40500 False False [0]
New samples 46500 False False [0]
New samples 63000 False False [0]
New samples 64500 False True [0]
Polling for first sequence for 2000 ms
Traceback (most recent call last):
  File "radar_control/radar_control.py", line 482, in <module>
    radar()
  File "radar_control/radar_control.py", line 470, in radar
    poll_timeout, seqnum_start + nave)
  File "radar_control/radar_control.py", line 278, in verify_completed_sequence
    raise ExperimentException(errmsg)
experiments.experiment_exception.ExperimentException: No first ack from driver - This shouldn't happen
`

kkotyk, 2018-01-16:
I then deleted the /dev/shm/ entries that were not from 'pulse' and it didn't have this error anymore.
I use the line:
`cd /dev/shm/ ; ls -1 | grep -v pulse | xargs rm -v` 
To remove the shm entries that failed to be removed by borealis.

kevinkrieger, 2018-01-18:
The error statement should be changed since, in fact, it CAN happen. Perhaps we add a suggestion to look at the shm entries and that the user has not filled up the partition?

mardet987, 2019-01-16:
outdated; communication redesign with brian

closed 2019-01-16 by mardet987