SuperDARNCanada/borealis #32: GPS lock wait time

kkotyk, 2018-10-19:
Currrent system will hang until gps is locked. We will need to figure out how we handle loss of GPS lock.

kkotyk, 2018-11-14:
Also, need to add clock address to config file and update in usrp.cpp

kevinkrieger, 2020-04-14:
Current status: We wait for GPS lock from the Octoclock-g in driver init. Once an GPS lock occurs, the USRP's time registers are all set to GPS time.  No checking of GPS lock is done after this to make sure GPS is still locked.

Ideal situation: Check GPS lock periodically, and have a field in data files that indicates if it is locked.
Or have a field in the data files that indicate the timing accuracy of our timestamps. If GPS is not locked, use the computer's time (disciplined by NTP) to set the USRP time registers. NTP can be queried for the estimated clock accuracy.

closed 2024-04-23 by tjk584