SuperDARNCanada/borealis #50: Noise.search and noise.mean fields in iqdat to be filled

mardet987, 2018-11-08:
Thinking that we could use a separate slice to be an rxonly slice for experiments that want noise information before transmitting (and/or to update transmit frequency based on noise info), and then this receive-only scan can be used within their own code to do an FFT and choose a frequency (using experiment update method). 

Pasha has mentioned that in acfs (rawacf and further data products), the noise value from the clear freq search is not really useful in comparison to the errors determined through averaging. At this point for backwards compatibility we will just place a zero in these fields but further work should be done on this

mardet987, 2019-04-29:
Noise.search is a useful background noise value for engineering purposes. More samples at the start of the pulse sequence could be added to calculate this background power level at the desired frequency with desired resolution bandwidth.
Noise.mean however, is not useful (mean of all power levels across the FFT). What could be useful instead is just storing the entire FFT of the background noise for the whole input rx bandwidth.
An FFT method needs to be built into signal processing methods.

mardet987, 2019-04-29:
I'm adding a noise_at_freq field (list of floats) into datawrite and adding this field into processeddata.proto for later fill

mardet987, 2020-04-03:
@kevinkrieger if you are working on clearfreqsearch code (and and FFT), you could also work on getting the noise fields filled in the data files. The noise_at_freq field in datawrite is currently just filled with zeroes.

kevinkrieger, 2020-04-14:
We can potentially add into the rawacf hdf5 files an entire FFT of a clr freq search band. If we use a 512 point FFT, and store only one byte for the amplitude of the noise power at each point, and only add one FFT per beam (integration period) then for every two hour file we would have approximately the following additional storage:

512 bytes/beam * 16 beams/scan * 120 scans/file = 983,040 bytes/file or approx 1MB extra at the standard number of beams in a file.

Then, the DMAP versions of the files' noise.search fields can be updated based on the values in the hdf5 files.

mardet987, 2020-07-07:
from issue #208 - as part of this issue add prints for noise and frequency chosen in radar_control when running