SuperDARNCanada/borealis #72: Experiment ID : What is it good for?

kkotyk, 2018-12-12:
Need to decide what to use for experiment ID.
Ideas thrown around: CPID, hash of the experiment

mardet987, 2019-04-23:
make it a hash of experiment_name

mardet987, 2020-04-14:
borealis_git_hash field contains the hash of the running Borealis commit. This could be used to find the experiment file that was running the name of this experiment if desired. CPIDs are still a useful number since all users are familiar with them.

mardet987, 2020-09-30:
Reopened in light of current discussions on ensuring uniqueness (and correctness) of CPIDs. @kkotyk suggested using the borealis_git_hash (experiment name would also be required and would need to be ensured unique) to identify the correct CPID for converted data after running, to remove the requirement for CPIDs in the experiment file. We should identify a database to refer to either way.