SuperDARNCanada/borealis #113: AGC status and Lo power inputs

mardet987, 2019-04-23:
should they only be monitored for logging or should they be written to file?

kevinkrieger, 2019-04-24:
I'm thinking monitored for logging, otherwise there will be many useless lines in the file. Kinda like an alert.

mardet987, 2020-04-13:
@kevinkrieger should this be closed with the recent merge of that PR for GPIOs?

kevinkrieger, 2020-04-13:
No, that PR just added some support for the test mode signal. I believe we still need to log every motherboard's agc, lopwr status.
Also, I have a better idea than just logging to file, we can utilize the agc_stat and lopwr_stat fields in the data files. This shouldn't be  a very large code change to the driver to loop over all motherboards (N200s). But to populate the agc_stat and lopwr_stat fields may involve more protobuf work and message parsing.

closed 2024-04-23 by tjk584