SuperDARNCanada/borealis #187: Removing an antenna causes a segfault in protobuf when trying to add the pin reads

kkotyk, 2020-04-20:


kkotyk, 2020-09-02:
@kevinkrieger @mardet987 do you guys remember what conditions we used that cause this to fail? I have different issues if I remove channels from the config, but it seems to work if I remove antennas from the experiment as intended.

mardet987, 2020-09-11:
> @kevinkrieger @mardet987 do you guys remember what conditions we used that cause this to fail? I have different issues if I remove channels from the config, but it seems to work if I remove antennas from the experiment as intended.

Not sure if you got an answer for this, but I'm pretty sure the conditions were when one of the USRPs addresses could not be found. Removing the antenna in the experiment will work fine if the USRP is not the problem (all USRP addrs are found). I think the usrp_driver needs to find all addresses listed in the config to run, regardless of which antennas are specified; so when it doesn't find one the only option is to remove from config at which time all the antenna numbers can get mixed up

mardet987, 2020-09-11:
Actually, I'm confused by the name of this issue; I may be talking about a different issue

tjk584, 2024-04-23:
Obsolete by #404

closed 2024-04-23 by tjk584