SuperDARNCanada/borealis #189: 20060101 date intermittent

mardet987, 2020-04-21:
I've seen a couple of times now where the radar will start writing data with date 20060101. Not sure why the timing gets reverted to this date. Goes a few hours writing with these dates before getting correct timing and writing with proper timestamps. The latest this happened was on March 28 2020 at PGR.

kevinkrieger, 2020-04-21:
Something to look into would be the GPS nmea sentences put out by this upon startup or after a GPS unlock:
https://datasheet.octopart.com/M9107-MtronPTI-datasheet-11562150.pdf

This person is experiencing the same thing: 
http://ettus.80997.x6.nabble.com/USRP-users-Octoclock-reference-unlock-td9186.html

You'll notice the timestamp is January 1st 2006 and his GPS refernce is unlocked.
The octoclock-g schematic: https://files.ettus.com/schematics/octoclock/octoclock.pdf

A potential fix for this would be to keep checking the GPS lock status in the driver, and sanity check the timestamp. This is tied with issue https://github.com/SuperDARNCanada/borealis/issues/32 and issue https://github.com/SuperDARNCanada/borealis/issues/52

tjk584, 2024-04-23:
We added a check to the `rsync_to_nas` data flow script to catch any files that have a timestamp inconsistent to the time they were created. See https://github.com/SuperDARNCanada/data_flow/pull/45