SuperDARNCanada/borealis #195: Transmit waveforms near edge of transmit bandwidth need characterization

kevinkrieger, 2020-04-23:
The transmit waveforms appear to be attenuated, have some modulation and are a different frequency (measured on oscilloscope) when attempting to transmit a frequency near the edge of the available transmit bandwidth.

Minimal testing has been done, but at least the following situation looked unacceptable:
Centre frequency: 12268.7500 kHz
TX bandwidth: 5.0e6 Hz
Lowest theoretical tx freq: 9768.7500 kHz
Highest theoretical tx freq: 14768.7500 kHz

Transmit frequency of 10.125MHz was unacceptable. See photo:

[image]

Transmit frequencies of 10.5, 11, 11.5, 13.025, 13.5 and 14 MHz all looked appropriate.

Transmit frequency 14.5MHz was unacceptable. See photo:

[image]

After changing the tx bandwidth (and rx bandwidth) to 6.25 MHz, this problem went away.
Because of this, I suspect this is caused by the rolloff of a filter in the DUC (or interpolator?) in the FPGA, and not Borealis code.

We'll need to characterize this rolloff, and increase our bandwidth to avoid frequencies near the edges, or just stay far enough away from the edges.

For now, staying 750kHz from the edge of the band seems to work well.

mardet987, 2020-04-23:
we can add a check for this in experiment_prototype and adjust TX bandwidth in experiment init

Doreban, 2024-03-28:
Transition bandwidth discussion came up again when refactoring borealis to enable re-tuning of the center tx and rx frequencies. This should be investigated for both transmit and receive bandwidth