SuperDARNCanada/borealis #199: short downtime source; requesting mem spans more than PCI bus

mardet987, 2020-05-20:
I've only seen this at Prince George site to date. It appears to be linked to short (I've seen 5 min or less) downtimes, from which the system recovers. 

> 2020-04-21T22:27:02.694105+00:00 borealis kernel: [929640.179996] caller _nv000906rm+0x1bf/0x1f0 [nvidia] mapping multiple BARs
2020-04-21T22:27:03.658093+00:00 borealis kernel: [929641.143609] resource sanity check: requesting [mem 0x000c0000-0x000fffff], which spans more than PCI Bus 0000:00 [mem 0x000c4000-0x000c7fff window]
2020-04-21T22:27:03.658103+00:00 borealis kernel: [929641.143743] caller _nv000906rm+0x1bf/0x1f0 [nvidia] mapping multiple BARs
2020-04-21T22:27:04.642095+00:00 borealis kernel: [929642.126087] resource sanity check: requesting [mem 0x000c0000-0x000fffff], which spans more than PCI Bus 0000:00 [mem 0x000c4000-0x000c7fff window]

mardet987, 2020-05-20:
On 20200425 at ~04:12, this occurs along with a lot of logs from gdm-x-session, followed by:

kevinkrieger, 2020-05-20:
Could this be related to the log-out issues we had that would kill the radar? I believe there were X messages in the log file related to that as well.