SuperDARNCanada/borealis #203: dependence on realtime.py to remove individual record files

kevinkrieger, 2020-06-11:
If realtime server doesn't reach any of the 'os.remove(filename)' lines in realtime.py's get_temp_file_from_datawrite() method, then it appears to affect the number of sequences sent within an averaging period. This seems to happen when realtime.py fails to run, and the rest of Borealis quietly continues on without informing us.

If possible, we should remove dependence upon the realtime server to remove these files, perhaps we can have a failsafe in data_write.py to remove older record files, or to make sure that realtime is running.

kevinkrieger, 2020-06-11:
the install_realtime() method in the install_radar_deps didn't execute properly when I installed Borealis, as that is where both pydarn and zmq are installed into the realtime venv

`rt_cmd = "cd /usr/local;" \
    "git clone https://github.com/vtsuperdarn/hdw.dat.git;" \
    "mkdir $BOREALISPATH/borealisrt_env;" \
    "virtualenv $BOREALISPATH/borealisrt_env;" \
    "source $BOREALISPATH/borealisrt_env/bin/activate;" \
    "pip install zmq;" \
    "pip install git+git://github.com/SuperDARNCanada/backscatter.git#egg=backscatter;" \
    "pip install pydarn;" \
    "deactivate;"
    execute_cmd(rt_cmd)
`

`kevin@Jupiter:~/borealis$ source borealisrt_env/bin/activate
(borealisrt_env) kevin@Jupiter:~/borealis$ pip install zmq
Traceback (most recent call last):
  File "/home/kevin/borealis/borealisrt_env/bin/pip", line 5, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'
`

RemingtonRohel, 2024-04-10:
This has been made obsolete in the develop branch by using h5py to write HDF5 files, and sending pickled dicts to realtime.py instead of relying on it to open and delete temporary files.

closed 2024-04-10 by RemingtonRohel