SuperDARNCanada/borealis #206: Review 60us TR window around transmit pulse

kevinkrieger, 2020-06-16:
We should review the hardware to determine how fast it can switch from transmit to receive and back. This will dictate how large of a window around the transmit pulse we need. 60us is the current value.

It may not be symmetric.

Doreban, 2024-04-02:
I've measured the HPTR response on switching.

For switching from receive to transmit, we get the following response:

[image]

The input signal (1: yellow) is entering the TX connector on the HPTR and the output signal (2: blue) is the output measured from the antenna connector. Signal 3: purple is the TR signal. It takes about 14us to relax, well within the current 60us window. As long as the window is larger than 20us the HPTR should be switched by the time transmission starts.

For switching from transmit to receive, we get the following response:

[image]

This switching noise occurs after the 60us buffer at the end of the transmit signal, so I don't believe we are restricted much on when this switch happens (at least when it comes to the HPTR). It is worth noting that this switching noise mostly settles after 20us like when switching from Rx to Tx. However, there is a DC offset on the signal that takes another 60us to fully settle out, so after switching to Rx there is an 80us window were switching noise might be noted. 

Rem has some raw RF data that shows the pulses and the the switching noise, so I'll grab that data from him and add it here tomorrow. As a quick summary in the mean time, it shows that the noise on switching to Rx happens far before any signal starts being recorded after the transmission and is very low power. So, other hardware components are the impacting factor on the TR switch to receive, and the HPTR noise doesn't impact the measurements much on the RX switch.

The switch to Tx has much stronger power and makes a much more notable impact on the measured samples.

Doreban, 2024-04-04:
Here is some data from a rawrf file when transmitting at 10.5 MHz at this time, with a center frequency of 12.0 MHz. The spike between 0.0290 and 0.0291 s is the TR switching noise to transmit. there is a delay of 60us then the pulse is transmitted. When the pulse ends, there is another 60us delay, then a smaller amount of noise from the TR switch to receive. There is then another ~40us delay before regular samples appear. So it seems like there is some hardware factors that further delay the rx path from being sampled right after the TR switch to receive is triggered.

[image]