SuperDARNCanada/borealis #236: datawrite writing multiple records' data arrays to single record

mardet987, 2020-10-08:
This is a somewhat common occurrence (~ few times per month at each site) where inside a file there will be a record where num_sequences stored != the num_sequences dimension in the data. The data array actually appears to have two records worth of data inside the single record which accounts for the issue. 

Most of the time this will happen in just one or two records in the file and then recover with no other files having this error.

However, a particularly bad case was 20200822/20200823 at Clyde River, where this occurred many times inside the same file. It may be related to realtime failing and not removing temporary files because this was the case in this instance (due to a necessary pydarn update not being yet implemented there). Another effect of this appeared to be that the sequences transmitted decreased to 1-2 per integration time.

kevinkrieger, 2020-10-08:
Potential leads to follow for this problem:
Assumptions: 
1) /dev/shm filling up and realtime not running are related somehow
2) realtime deletes the temp files in the data directory, data_write deletes the shm files after all data is written

data_write.py:

line ~671 :`so.send_data(rt_dw['socket'], rt_dw['iden'], tmp_file)`

Does this call to send_data(...) block? 

This call is ultimately a call to zmq's send_multipart(...) - which calls zmq's send(...) with the SNDMORE flag until it's done sending all the parts except the last, at which time it doesn't have the SDNMORE flag set. 

ZMQ documentation says the following about DEALER sockets, which the rt_dw socket is:
> When a ZMQ_DEALER socket enters an exceptional state due to having reached the high water mark for all peers, or if there 
> are no peers at all, then any zmq_send(3) operations on the socket shall block until the exceptional state ends or at least one 
> peer becomes available for sending; messages are not discarded.

The High Water Mark (HWM) is default set to 0, which means 'no limit'. (see here: http://api.zeromq.org/2-1:zmq-setsockopt)
Questions this documentation brings up: 
1) Are there any peers for this socket when realtime is not running? 
2) Does Brian act as the peer?
3) If the call is actually blocking, then why do we get 1 or 2 averages? 

Lines ~969-973 close and delete the shared memory after writing raw rf parameters to the data file (this only happens if 'write_raw_rf' is set):
```
# Can only close mapped memory after its been written to disk.
for shm, mapfile in zip(shms, mapfiles):
    shm.close_fd()
    shm.unlink()
    mapfile.close()
```

lines ~1109-1112 close and delete the shared memory after writing all file types except 'write_tx', this only happens if 'write_raw_rf' is not set.
```
for rf_samples_location in data_parsing.rawrf_locations:
    shm = ipc.SharedMemory(rf_samples_location)
    shm.close_fd()
    shm.unlink()
```

kevinkrieger, 2020-10-08:
There's some discussion that is closely related here: https://stackoverflow.com/questions/27107066/how-do-you-tell-if-zmq-socket-send-multipart-fails-with-dealer-router-socket-pai

RemingtonRohel, 2022-04-08:
Related discussion: https://github.com/SuperDARNCanada/borealis/issues/301 and https://github.com/SuperDARNCanada/borealis/issues/263

closed 2022-08-30 by RemingtonRohel