SuperDARNCanada/borealis #239: scons build on Ubuntu missing pthread_condattr_setclock@@GLIBC_2.3.3

alexchartier, 2020-10-23:
I went through the build instructions on Ubuntu and got as far as scons, but got:
undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3' 

Do you know how to correct that? Is it a GLIBC problem, or a makefile issue?

alexchartier, 2020-10-23:
PS: the -lpthread option is all that it needs to compile that line:
g++ -o build/release/tools/rx_tx_delay_test/rx_tx_delay_test build/release/tools/rx_tx_delay_test/rx_tx_delay_test.o -luhd -lboost_system -lboost_thread -lboost_program_options -lpthread

So, the question is how to get that into the compiler. Should it be in boost_program_options?

alexchartier, 2020-10-23:
Wait...
we added 'pthread' to the list of libraries in the Sconscript t of 'rx_tx_delay_test' and that fixed it

A few modules are different in Ubuntu vs OpenSUSE install dependencies

alexchartier, 2020-10-23:
Let me know if there's a downside to doing that and if not I'll close the issue

kkotyk, 2020-10-29:
I don't see an issue adding that dependency. The reason I picked scons for building was for how easy it is to add stuff like this.

closed 2020-11-05 by alexchartier