SuperDARNCanada/borealis #240: NTP sync

alexchartier, 2020-10-23:
Why do we have to do the timeserver stuff? 

"timedatectl status " says NTP service is active and the system clock is synchronized. 

The test (Step 2 of current instructions for verifying PPS) looks ok until "fetching data" at which point it says no response. Presumably we don't have access to the USask time server specified earlier.

alexchartier, 2020-10-23:
[image]

alexchartier, 2020-10-23:
[image]

kevinkrieger, 2020-10-29:
Hey Alex, You don't have to set up NTP for Borealis to run, but it's easy to do and your OS will be as close as possible to GPS time. I would find an alternative NTP server (I'm guessing JHU-APL has multiple considering this: https://tools.ietf.org/wg/ntp/draft-ietf-ntp-ntpv4-algorithms/draft-ietf-ntp-ntpv4-algorithms-01-from-00.diff.txt), the documentation is just an example that works for us. There's also this list of public servers, some of which are near you: https://tf.nist.gov/tf-cgi/servers.cgi

I suspect you're getting that error because the PPS signal cable either isn't connected to your motherboard's COM port, or the kernel module to get PPS signals from the COM port isn't loaded. This is something I realize we have not documented (how to make/install the cable) so I wouldn't worry about it for now. If you set up NTP with a timeserver from the above public list, that will get you most of the way there. Once we make some documentation for the PPS, that just increases the NTP accuracy.

Dog tax approved!  :100:

kkotyk, 2020-10-29:
There are some secondary functions of the Borealis computer that don't directly use the USRP GPS timing(ex. scheduling). Although the time keeping software in linux is miles ahead of that in QNX, we figured we would discipline the OS time since we have extra PPS outputs. Some of the stuff we've included is an attempt to minimize/eliminate sources of error that have caused problems with in the ROS system.

closed 2020-11-04 by kevinkrieger

JWiker, 2020-11-09:
@kevinkrieger when you say _The Borealis computers at several SuperDARN Canada sites are disciplined to within a few microseconds of true GPS time_, do you mean UTC via GPS or real GPS time?  I just wanted to make sure since there's currently a difference of 18 seconds between GPS time and UTC since GPS time does not include leap seconds