SuperDARNCanada/borealis #245: Possible indexing issue when plotting antennas_iq

kevinkrieger, 2020-11-17:
There seems to be an issue when plotting antennas_iq.
I've run the code as follows:

`cp /data/borealis_data/20200612/20200612.0000.00.sas.0.antennas_iq.hdf5.site /home/radar/`

`source ~/pydarn-env/bin/activate`

`python3 ~/data_flow/site-linux/borealis_convert_file.py 
/20200612.0000.00.sas.0.antennas_iq.hdf5.site`

`python3 ~/borealis-data-utils/SNR/plot_antennas_range_time.py ~/20200612.0000.00.sas.0.antennas_iq.hdf5`

I'm sorry, not sure why github is splitting these up, I can't seem to remove the newlines in the python3 calls.

Anyways, this results in some divide-by-zero errors for the last few antennas, and the plots appear to be incorrect, specifically for the last two antennas:

> (pydarn-env) radar@sasborealis:~> python3 ~/borealis-data-utils/SNR/plot_antennas_range_time.py ~/20200612.0000.00.sas.0.antennas_iq.hdf5
> /home/radar/20200612.0000.00.sas.0.antennas_iq.hdf5
> antenna_0
> antenna_1
> antenna_2
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_0_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_1_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_2_0_70.png
> antenna_3
> antenna_4
> antenna_5
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_3_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_4_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_5_0_70.png
> antenna_6
> antenna_7
> antenna_8
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_6_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_7_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_8_0_70.png
> antenna_9
> antenna_10
> antenna_11
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_9_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_10_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_11_0_70.png
> antenna_12
> antenna_13
> antenna_14
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_12_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_14_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_13_0_70.png
> antenna_15
> antenna_16
> antenna_17
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_15_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_16_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_17_0_70.png
> antenna_18
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 102: RuntimeWarning: divide by zero encountered in log10
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 104: RuntimeWarning: divide by zero encountered in log10
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 107: RuntimeWarning: invalid value encountered in subtract
> antenna_19
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 102: RuntimeWarning: divide by zero encountered in log10
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 104: RuntimeWarning: divide by zero encountered in log10
> /home/radar/borealis-data-utils/SNR/hdf5_rtplot_utils.py: 107: RuntimeWarning: invalid value encountered in subtract
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_18_0_70.png
> /home/radar/20200612.0000.00.sas.0.antennas_iq.antenna_19_0_70.png

[image]
[image]

mardet987, 2020-11-17:
This appears to be a borealis mem copy issue with the last two antennas:

I searched for sequences where the entire sequence is zeros in this specific file (20200612.0000.00.sas.0.antennas_iq.hdf5):
>record 1 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 2 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 3 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 4 antenna 18 sequences [32, 33, 34, 35]
record 4 antennas [19]
record 5 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 6 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 7 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 8 antenna 18 sequences [32, 33, 34, 35]
record 8 antennas [19]
record 9 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 10 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 11 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 12 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 13 antenna 18 sequences [32, 33, 34, 35]
record 13 antennas [19]
record 14 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 15 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 17 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 18 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 19 antenna 18 sequences [32, 33, 34, 35]
record 19 antennas [19]
record 20 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 21 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 22 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 23 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 24 antenna 18 sequences [32, 33, 34, 35]
record 24 antennas [19]
record 25 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 26 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 27 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 28 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 29 antenna 18 sequences [32, 33, 34, 35]
record 29 antennas [19]
record 30 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 31 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 33 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 34 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 35 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 36 antenna 18 sequences [32, 33, 34, 35]
record 36 antennas [19]
record 37 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 38 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 39 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 40 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 41 antenna 18 sequences [32, 33, 34, 35]
record 41 antennas [19]
record 42 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 43 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 44 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 45 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 46 antenna 18 sequences [32, 33, 34, 35]
record 46 antennas [19]
record 47 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 49 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 50 antenna 19 sequences [16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35]
record 51 antenna 18 sequences [32, 33, 34, 35]
record 51 antennas [19]

There is a pattern here, although the number of records before antenna 18 has missing sequences varies from 3-5.

mardet987, 2020-11-17:
ah I just noticed that records 0, 16, 32, 48 do not have missing sequences for any antennas

mardet987, 2020-11-17:
I'll check more for this behaviour in recent files

mardet987, 2020-12-08:
this is a pydarn v1.0.0 issue; it was resolved with pydarn v1.1 (and any version of pydarnio)

closed 2020-12-08 by mardet987