SuperDARNCanada/borealis #359: install_radar_deps not working properly

Lozzy-Bear, 2022-09-15:
The following are problems encountered while using install_radar_deps.py from main on a fresh build.

- crashes and soft locks the computer when installing cupy
- does not install python3-Mako, PyYAML, or requests which crashes UHD install
- installing env are not tagged with python version which raises and error when running
- after running, cannot import /usr/local/hdw.dat
- config files appear to be an outdated version by default

After numerous failures, I installed everything manually, and it mostly worked.

RemingtonRohel, 2022-10-21:
Haven't installed Borealis fresh on a computer lately, but I can say the following:

* Config file repo has been updated since this comment
* Tested installation of virtual environments, seems to be working correctly.
* Haven't had issues installing cupy on several computers since - not sure why this was a problem
* Also haven't had issues with accessing hdw files - seems to be working now

Lozzy-Bear, 2023-04-20:
Recent updates have resolved this issue.

Lozzy-Bear, 2023-04-20:
This can be closed.

closed 2023-04-20 by Lozzy-Bear