SuperDARNCanada/borealis #408: Scheduler not starting radar successfully when no schedule available

tjk584, 2023-07-19:
Symptoms:
I manually stopped the radar at CLY to run some tests. When I tried to run it again using start_radar.sh, the radar wouldn't start successfully. I was able to run it manually using steamed_hams.py, but using the scheduler didn't work. At this time, CLY had no schedule available (due to network being down for past couple months).

Solution:
After syncing the new schedule from Campus to CLY, the radar started as normal using the scheduler.

What to do:
This problem should be looked into to make sure it's just a one off, and it doesn't prevent a site from operating without a schedule in the future. It should be noted that CLY ran for the past month with no schedule - not sure why it stopped working once I manually restarted the radar.

RemingtonRohel, 2024-04-10:
I bet this is related to the scheduler problems we have been having lately, see #448 and #447

RemingtonRohel, 2025-04-24:
This should be handled now, `remote_server.py` should always fall back to the most recent infinite schedule line.

closed 2025-04-24 by RemingtonRohel