SuperDARNCanada/borealis #409: Default frequency in restrict.dat files

RemingtonRohel, 2023-07-20:
What is the purpose of the default frequency line in the `restrict.dat.[site]` files? Some sites have the same default frequency. Currently it is impossible to run an experiment without specifying a frequency for each slice, which makes the default frequency a useless field. Unless we want to change it to the `COMMON_MODE_FREQ` from `superdarn_common_fields.py`?

JWiker, 2023-10-25:
@RemingtonRohel did you ever find an answer to this?

RemingtonRohel, 2023-10-25:
I did not. The field is read in from the file but never accessed afterwards, so it seems to me that it is unnecessary. However the code will throw an exception if default frequency isn't specified in the restrict.dat file. This could definitely be removed in a future release, it seems like a useless requirement to have.

closed 2026-08-28 by KieranKuch