SuperDARNCanada/borealis #422: Multiple instances of Borealis starting

tjk584, 2023-12-01:
While setting up Nagios on invmain206, I accidentally ran a `check_borealis` script that attempts to restart the radar. This script executed many times in quick succession, resulting in 7 different Borealis screens created at the same time. Luckily, this computer wasn't running the radar, so no downtime occurred.

This issue is somewhat unrelated, but we should put a check to make sure only one instance of Borealis can be started at a time. It shouldn't be possible to start 7+ Borealis processes at once - if one Borealis instance sees another already in operation, it should stop before starting another instance.

tjk584, 2024-04-10:
This problem occurred on rknmain207 yesterday. The radar was running (successfully? not sure), but ~10 Borealis screens were created and opened. Different modules were running in different screens.

rknmain207 had been down since January, so when it started back up it was running older software. Not sure what was causing this problem.

RemingtonRohel, 2024-10-29:
steamed_hams.py has a command (`screen -X -S borealis quit`) near the end to kill any Borealis screens already open, so I suspect this may have been a race condition from multiple steamed_hams invocations started in very short succession.

tjk584, 2024-10-29:
We could add a check to ensure any instance of steamed_hams is a singleton. Our data flow scripts do this (but in bash):

```bash
# Ensure that only a single instance of this script runs.
if pidof -o %PPID -x -- "$(basename -- $0)" > /dev/null; then
	printf "Error: Script $0 is already running. Exiting...\n"
	exit 1
fi
```