SuperDARNCanada/borealis #464: Borealis file timestamped Jan 1 1970

tjk584, 2024-04-24:
Borealis at Saskatoon recently produced a file timestamped 19700101.0000.00. The file was generated on April 20, 2024.

The file is available in `/borealis_nfs/borealis_data/conversion_failure/` at Saskatoon:

```
transfer@sasdist205:/borealis_nfs/borealis_data/conversion_failure> ls -l 19700101.0000.00.sas.1.*
-rwxrwxr-x 1 transfer users 1589768 Apr 20 13:37 19700101.0000.00.sas.1.antennas_iq.hdf5.site
-rwxrwxr-x 1 transfer users   52536 Apr 20 13:45 19700101.0000.00.sas.1.rawacf.hdf5.site
```

May be related to #189 #222

RemingtonRohel, 2024-05-15:
There was another file whose `sqn_timestamps` field was full of timestamps from 1970, even though the filename was in 2024. The filename is made from the first sequence timestamp received from the `usrp_driver.cpp` module, which gets the timestamp from the USRP time (`borealis_clocks.usrp_time = meta.time_spec`, where `meta` is the `rx_metadata_t` struct returned by a call to `recv()` on the `multi_usrp` object). All that to say, for some reason the `recv` call returned a timestamp that was only seconds past epoch, instead of millions of seconds past.

RemingtonRohel, 2024-07-31:
This could be caught by the radar before data is written to file, by inspecting the `gps_to_system_time_diff` field. The `usrp_driver.cpp` process creates the field originally, and could raise an error if the value is erroneously large (e.g. 1e9 for current the most recent 1970-timestamped files). This would crash the radar, and an automated restart would then likely rectify the discrepancy.

RemingtonRohel, 2024-10-18:
Closed by #497

Doreban, 2025-03-20:
A 1970 timestamped file appeared on the lab computer generated on 2025 March 18. The timestamps in the files were all 19700101.0000.21 and the time difference between the system and gps that was logged was 12 seconds. 

This issue has cropped up again so I am re-opening the issue.

Doreban, 2025-03-20:
Looking into it a bit more the current check for poor timestamps uses the `borealis_clocks` which is composed of a `uhd` time and a `chrono` time which is initialized just before the `samples_metadata` is filled.

The metadata is filled using the `sequence_start_time` which is defined from `initialization_time` and `time_from_initialization` which are pulled from the start of the receive thread and in the while loop of the receive thread. 

Its possible that the bad date could have been stored in the init and been resolved before `borealis_clocks` is set. It might be good to compare the system time directly to `sequence_start_time`.