SuperDARNCanada/borealis #536: New CFS implementation ideas

RemingtonRohel, 2025-06-09:
BAS/La Trobe and the American USRP systems use dead time between sequences to extract spectrum estimates. They run a separate server that uses a running-average spectrum estimate from these samples for CFS. They also don't beamform, instead using incoherent averaging of the spectrum across antennas. These are some good ideas for how to upgrade/modify our implementation for flexibility. We could add another screen to the steamed_hams window that shows the current spectrum estimate as a bar or line graph, that would be really cool.

Some considerations:
* Slowdown due to extra GPU usage for filtering the dead time samples and FFT-ing them.
* Estimates being obsolete if they were taken during an AVEPERIOD and the next time that beam runs is one minute later (e.g. normal operations, we care about the spectrum for the current beam but the last time we saw that beam was a minute ago).

RemingtonRohel, 2025-06-10:
Possible tool for FFT visualization: https://pypi.org/project/itrm/

Doreban, 2025-06-18:
Further on this. I think it would be beneficial to shift CFS into a background task that is "always running". During downtime between pulses collect data and keep a running track of the spectrum data. We could then:

* Have data write pull the CFS processes for the most recent spectrum data if we want to right it to file
* Stream the spectrum data over realtime
* Slices that want to use CFS to update a frequency they poll the CFS process similar to data write